ThingPulse/minigrafx #2: Rotation

mattncsu, 2017-06-30:
Is it possible to draw some text in a different rotation from the rest of the sketch? I tried
`gfx.setRotation(0);`
`gfx.drawString(10, 10, "Hello World");`
`gfx.setRotation(1);`
`gfx.drawString(20, 20, "Hello World");`
`gfx.commit;`
but rotating between drawString calls corrupted the buffer.

squix78, 2017-06-30:
I have rotated text on my backlog but could not implement it yet. Your rotation trick probably doesn't work because it only takes real effect once the buffer is written to the display.
I would either implement it with an additional setTextRotationMethod plus adapting the drawString method or by adding a overloaded drawString method with additional rotation parameter...
Let me know if you are interested in contributing this feature by pull request and we'll discuss the design...

mattncsu, 2017-06-30:
I'm can sure try to contribute although my abilities are mostly limited to hacking together code from different places.  I'd start with [this](https://github.com/loboris/ESP32_ePaper_example/blob/master/components/epaper/EPD.c) library written for the ESP32.  I did write an invert colors function for your library that I can try and figure out how to get it into MiniGrafx.cpp (It is just a function in EPD_WaveShare_2_9.ino right now) and make that my first official pull request.

mattncsu, 2017-07-06:
I have some basic functionality (https://github.com/squix78/minigrafx/pull/3/files) that works for some limited angles but usually ends up off screen because the text is spinning about some odd point.  Most of the functionality is in setPixel90().  Any suggestions?

closed 2019-06-03 by stale[bot]