ThingPulse/minigrafx #6: Big fonts for old eyes?

jumblies, 2017-07-26:
Thank you for the new library.  All your hard work is appreciated.  

Is it possible to get fonts around 72pt to work with this library?  

I created a 72pt Roboto Black font using [http://oleddisplay.squix.ch/#/home](url) but it ended up only displaying the left hand side of the numerals.  

44pt works great.  50 has some issues at the margins(time in the pic below) but is passable but 72 is mostly incomplete (the temp in the pic below.

[image]

[roboto black big fonts.zip](https://github.com/squix78/minigrafx/files/1177811/roboto.black.big.fonts.zip)

wawibu, 2017-09-29:
I'm running into the same issue that fonts in this side not shown correct. Getting the same error message and the same output on the screen. Would be great if there is a way to get this fixed :)

skx, 2018-06-20:
The problem is the generated code is creating numbers that are too large for the type.

If you look at the output of the [generating site](http://oleddisplay.squix.ch/#/home) you'll see this:

     const char Dialog_plain_78[] PROGMEM = {
	0x4E, // Width: 78
	0x5C, // Height: 92
	0x20, // First Char: 32
	0xE0, // Numbers of Chars: 224

	// Jump Table:
	0xFF, 0xFF, 0x00, 0x19,  // 32:65535
	0x00, 0x00, 0xEE, 0x1F,  // 33:0
	0x00, 0xEE, 0x13D, 0x24,  // 34:238
        ... more

So you see it is creating an array of `char`.  A `char` will hold values 0-0xFF (0-255).  But the values in the list are bigger than that.  e.g. `0x13D` is bigger than `0xFF`.

To fix this you need to do two things:

* Change the type:
* See if it works.

In the generated output change  `const char Dialog_plain_78[] PROGMEM = {` to:

      const int Dialog_plain_78[] PROGMEM = {

That will ensure those values are correct.  Of course the next step is seeing if what _consumes_ the fonts will work with `int` rather than `char`.  You'll either get lucky, or not.  If it fails to compile you'll soon spot the problem and I suspect it should be as simple as updating the types appropriately.  (i.e. Carrying on changing `char` to `int` as you see error-messages about truncation, or mis-matched types.)

Aljrob, 2018-10-02:
So I'm a bit new to c/c++ but followed skx's explanation above (thanks btw) and have updated my font type to int.  

Unfortunately the Minigrafx library itself does seem to be struggling to consume the int font data, with the compiler erroring out with the following in a few places in my code.

no matching function for call to 'MiniGrafx::setFont(const int [64671])'

Attempting to update the Minigrafx library is probably beyond my current skill level without guidance at the moment - has anyone managed to successfully update Minigrafx to show large fonts?

xriser, 2019-04-02:
Same issue, please fix it, who can.
Thank you in advance

closed 2019-10-13 by stale[bot]