ThingPulse/minigrafx #9: Share the font generator code

stojanos, 2017-10-09:
Would it be possible to share font generation code ? 

Amy help in getting fonts generated would be great (e..g how does it compare to Adafruit fonts, would it be easier to convert Adafruti GFX fonts etc) 

We would also be able to improve the code if required.

Thanks,
Srdjan

stojanos, 2017-10-12:
Just to inform you. I've found Java app on some of your previous posts, downloaded and generated my fonts without any problems.
Thanks for creating such a neat library. If only we would have microcontrollers with more RAM to use a full potential of this library.
Cheers!!!

mattncsu, 2017-10-12:
Do you mind linking to one of the posts?

stojanos, 2017-10-12:
Can't find the post any more, but I believe code is here:
https://github.com/squix78/esp8266-oled-ssd1306-font-converter

Simple:
- download  this single java file,
https://github.com/squix78/esp8266-oled-ssd1306-font-converter/blob/master/src/main/java/ch/squix/esp8266/fontconverter/rest/FontConverterV3.java
- create project in your favorite Java IDE (NetBeans IDE standard will do)
- inside the code set the name of the font, size and start and end char (note last char will nto be exported so always use one more)  font must be installed on the system - works on both windows and ubuntu
- run app - header file will be printed to system output.
Enjoy !

mattncsu, 2017-10-15:
Awesome thanks!

lorol, 2017-11-06:
@stojanos , thank you for the hint!
[Here](https://github.com/lorol/esp8266-oled-ssd1306/blob/master/resources/glyphEditor.html) you can find a standalone HTML+JS "visual editor", originally made by [Reivaxy](https://github.com/reivaxy/esp8266-oled-ssd1306). I just added some stuff to it. This tool allows you load and see/edit the generated C code by a browser like Mozilla.

closed 2019-06-03 by stale[bot]