ThingPulse/minigrafx #10: Too less memory

MarcusF2015, 2017-10-20:
I tried to expand the color weatherstation with using Mozilla Location Service MLS and json-streaming-parser. MLS needs a clientsecure (https) connection. 
minigrafx, clientsecure and the String class need too much RAM, causing an exception 29 crash.
Is it possible to initialisize minigrafx after closing clientsecure ?
So I would like to geolocate in setup to get the coordinates before using minigrafx.

Marcus

MarcusF2015, 2017-10-21:
I found changeBitDepth to reduce need of memory. Is it possible to expand changeBitDepth to 0 bit to do memory hungry WiFiClientSecure without display? After the HTTPS transmission the display shoud work againg with changeBitDepth 1, 2, ... bit.

Marcus

ht81, 2018-06-13:
I run into same problem due to the error expection 29. i am using mqtt with clientsecure https. any workaround?

closed 2021-06-26 by stale[bot]