ThingPulse/minigrafx #13: fillBuffer won't work as expected;

cjbrigato, 2017-11-26:
Hello,

The fillbuffer function will not work as expected as is :
```
void MiniGrafx::fillBuffer(uint8_t pal) {
    uint8_t byteCol = pal;
    for (int i = 0; i < pixelsPerByte; i++) {
      byteCol = byteCol | (pal << i);
    }
    memset(buffer, byteCol, bufferSize);
}
```
Saw it when I tried to use another palette from your standard one.
You can try it with such palette (standard 16-color with shades of grey, classic indexed color's one) : 
```
uint16_t color565(uint8_t r, uint8_t g, uint8_t b)
{
  return ((r & 0xF8) << 8) | ((g & 0xFC) << 3) | (b >> 3);
}

uint16_t palette[16] = { 
  ILI9341_WHITE, color565(0x00,0xFF,0xFF), color565(0xFF,0x00,0xFF), color565(0x00,0x00,0xFF),
  color565(0xFF,0xFF,0x00), color565(0x00,0xFF,0x00), color565(0xFF,0x00,0x00), ILI9341_BLACK,
  color565(0xDE,0xDE,0xDE), color565(0xBF,0xBF,0xBF), color565(0x9F,0x9F,0x9F), color565(0x7F,0x7F,0x7F),
  color565(0x62,0x62,0x62), color565(0x4B,0x4B,0x4B), color565(0x3A,0x3A,0x3A), color565(0x2E,0x2E,0x2E)
}
```

Just try to fillbuffer with black : 
```
gfx.fillBuffer(7);
gfx.commit();
```
It will result in a whiteish screen. This because as smart as was thought the byteCol bitwise operation before the memset, it just doesn't work as is.

So right now at least to 4bits per pixel palette, a working fillBuffer function would be : 

```
void MiniGrafx::fillBuffer(uint8_t pal) {
    uint8_t byteCol = pal;
    byteCol = byteCol | (pal << bitsPerPixel);
    memset(buffer, byteCol, bufferSize);
}
```

Even better would be to rewrite it functionnal from 1 to 8 bit per pixel but I guess a lot of us use it a 4 or less.

spilz87, 2018-10-30:
the issue is from pal<<i
the right function is :

void MiniGrafx::fillBuffer(uint8_t pal) {
    uint8_t byteCol = pal;
    for (int i = 1; i < pixelsPerByte; i++) {
      byteCol |=  (pal << (i*bitsPerPixel));
    }
    memset(buffer, byteCol, bufferSize);
}

cjbrigato, 2018-12-01:
I was sure I had updated some days later this issue by adding back the pixelsPerByte iteration... sorry.

Thanks you, closable I guess :)

closed 2019-06-13 by stale[bot]