ThingPulse/minigrafx #21: problem using truetype font (created with fontconvert)

glown, 2018-03-04:
I have a couple of fonts created with adafruit's fontconvert and I cannot get them to compile.  I'll continue to debug, but wanted to verify that the idea is ok:

Should a font created from truetype using adafruit fontconvert work with minigrafx?

glown, 2018-03-05:
more information .... newfont36 was created with adafruit fontconvert program and has worked with other libraries.  I'd like to take advantage of the display buffer of minigrafx to reduce flicker or the need to clear the background before printing to the screen.

when I add the following command:
 gfx.setFont(&newfont36pt7b);

I get the following compile errors:
MiniGrafx.h:127:8: note: void MiniGrafx::setFont(const char*)
   void setFont(const char *fontData);
        ^
C:\Users\winusr\Documents\Arduino\libraries\Mini_Grafx\src/MiniGrafx.h:127:8: note:   no known conversion for argument 1 from 'const GFXfont*' to 'const char*'
exit status 1
no matching function for call to 'MiniGrafx::setFont(const GFXfont*)'

squix78, 2018-03-05:
I think you have to change the data type in the created font file. The setFont method has the signature "void setFont(const char *fontData);" and you are trying to feed it a const GFXfont reference. The compatibility towards Adafruit generated fonts is only on the data level, not on the data type level... Have a look what our font generator generates when you choose Adafruit GFX font: http://oleddisplay.squix.ch/#/home

glown, 2018-03-05:
I have not been able to get a gfx font to compile, even after trying to change the data type in the font file.  However, a font created using Library Version >= 3.0.0 on oleddisplay.squix.ch compiles fine.  

Is the font converter code for library version >=3.0.0 available in github?  I thought I was in the pink when I found adafruit's gfx fontconvert program since I could create any size fonts from my truetype original.  But now I need an equivalent fontconvert that generates library version >=3.0.0 files.

squix78, 2018-03-05:
Can you please paste the first few lines of your font file here? It's hard
to help you on the blind...

glown, 2018-03-05:
For testing I've concentrated on fonts generated from oleddisplay.  a version >=3.0.0.0 and a gfx version.

First few lines of >=3.0.0.0 (**this compiles**)
// Created by http://oleddisplay.squix.ch/ Consider a donation
// In case of problems make sure that you are using the font file with the correct version!
const uint8_t DejaVu_Sans_16Bitmaps[] PROGMEM = {
	// Bitmap Data:
	0x00, // ' '
	0xFF,0x30, // '!'

squix78, 2018-03-06:
So just to be clear, changing the first line of the GFX fontconvert output to
const char DejaVu_Sans_16Bitmaps[] PROGMEM = {
// Bitmap Data:
0x00, // ' '
0xFF,0x30, // '!'
doesn't work? What is the compile error? I would understand that you are getting a runtime error, but a compile error?

glown, 2018-03-06:
That is correct. Changing the first line to char results is the following compile error:

C:\Users\winusr\Documents\Arduino\v02-MiniGrafx\v02-MiniGrafx.ino: In function 'void loop()':
v02-MiniGrafx:61: error: no matching function for call to 'MiniGrafx::setFont(const GFXfont&)'
   gfx.setFont(DejaVu_Sans_16);
                             ^
C:\Users\winusr\Documents\Arduino\v02-MiniGrafx\v02-MiniGrafx.ino:61:29: note: candidate is:
In file included from C:\Users\winusr\Documents\Arduino\v02-MiniGrafx\v02-MiniGrafx.ino:3:0:
C:\Users\winusr\Documents\Arduino\libraries\Mini_Grafx\src/MiniGrafx.h:127:8: note: void MiniGrafx::setFont(const char*)
   void setFont(const char *fontData);
        ^
C:\Users\winusr\Documents\Arduino\libraries\Mini_Grafx\src/MiniGrafx.h:127:8: note:   no known conversion for argument 1 from 'const GFXfont' to 'const char*'
exit status 1
no matching function for call to 'MiniGrafx::setFont(const GFXfont&)'

After that if I additionally change the last line to char like this:

const char DejaVu_Sans_16 PROGMEM = {
(uint8_t  *)DejaVu_Sans_16Bitmaps,(GFXglyph *)DejaVu_Sans_16Glyphs,0x20, 0x7E, 19};

I get the following compile error:
In file included from C:\Users\winusr\Documents\Arduino\v02-MiniGrafx\v02-MiniGrafx.ino:6:0:
gglx.h:199: error: scalar object 'DejaVu_Sans_16' requires one element in initializer
 (uint8_t  *)DejaVu_Sans_16Bitmaps,(GFXglyph *)DejaVu_Sans_16Glyphs,0x20, 0x7E, 19};
                                                                                 ^
exit status 1
scalar object 'DejaVu_Sans_16' requires one element in initializer

squix78, 2018-03-06:
The first error ( error: no matching function for call to
'MiniGrafx::setFont(const GFXfont&)') strongly indicates that you didn't
actually change the type of the gfx font to const char. Please zip your
whole project and attach it here...

glown, 2018-03-06:
It's possible I've misunderstood.  Here's the project I'm using:
[v02-MiniGrafx.zip](https://github.com/ThingPulse/minigrafx/files/1785659/v02-MiniGrafx.zip)

closed 2019-06-03 by stale[bot]