ThingPulse/minigrafx #22: examples (RotatingCube OK, Carousel shows no graphics)

taogeek, 2018-03-25:
When running the carousel demo for il9341 the screen initializes (no content) and then ... nothing.  The program is running (I can print the value of remainingTimeBudget  in the main loop).  With the same hardware I can run the rotatingcube demo and get a very nice greaphical demo on my display. 

I have a il9341 tft lcd without touch pins attached (for the moment) driven with a to a wemos d1 mini.  I tried a nodemnu also, and it behaved the same.

Am I missing something incredibly basic?  I wanted to learn about the carousel funtionaliaty and thought the demo was the beat place to start.  I should "see" something???

glown, 2018-03-26:
I ran the carousel example too and nothing displayed.  Weird.  Please add a comment here if you figure out the problem.

spilz87, 2018-10-31:
it's just because carousel doesn't clear and display itself, it's a part of the screen, to run the exemple code modify just like this : 
`void loop() {
  gfx.fillBuffer(0);
  int remainingTimeBudget = carousel.update();
  gfx.commit();
  
  if (remainingTimeBudget > 0) {
    // You can do some work here
    // Don't do stuff if you are below your
    // time budget.
    delay(remainingTimeBudget);
  }
}`

closed 2019-06-03 by stale[bot]