ThingPulse/minigrafx #24: mini_grafx issue with latest ESP32 support

hsgentry, 2018-04-21:
I have used mini_grafx successfully for many ESP8266 and ESP32 projects.   However, since I upgraded to the latest support package for ESP32 the display occasionally goes all white with nothing else displayed.  I have tested this with the exact same software on the exact same hardware with the only difference being the ESP32 package.  The one from a couple months ago works perfectly.  The one recently posted works fails in anything from 10 minutes to many hours.  The processor is still running with no faults.  I can see from Serial prints that everything is working, including the touch screen.

I don't think this is actually a mini_grafx  issue, but I suppose it could be.  It could be something that I am doing in the code, but then it works perfectly with the older ESP32 support.

hsgentry, 2018-05-01:
I have traced this issue to the XPT2046_Touchscreen library. If isTouched() is not called then the display will work perfectly.  Since the touch and display share the SPI I am guessing that there is some issue with the latest ESP32 support relating to more than one device on SPI.

Donderda, 2018-05-29:
Could you use another SPI (i.e. HSPI) for your Touchscreen?

hsgentry, 2018-05-30:
That was the first thing I tried, but could not make it work at all.  I
must have made a mistake when I tested it because I just tried again today
and it seems to be running.  Time will tell if this fixes the problem.
However, if it does then it just fixes my problem.  I still think there is
something wrong with the latest ESP32 support (maybe in the SPI library)
that makes it unreliable if there are two devices on one bus.  The version
from November worked perfectly with two devices on one SPI bus.  It could
easily go unnoticed because it will sometimes run for several days before
it fails.

Sam

hsgentry, 2018-05-30:
Well... I think I am ready to give up.  I modified the touch library to use
a second SPI bus and it worked just fine with the
"hardware/esspresif/esp32" from November 2017, but does not work at all
with the version from April 2018.  So, the latest ESP32 arduino package
does not work with two devices on one bus and does not work with two SPI
busses.  I have no choice but to use the older ESP32 package.

I suppose I may be doing something wrong, but it is relatively easy to do.
I just:

added:
SPIClass SPI2(VSPI);

then:
SPI2.begin(17,16,4,5);

and change all the SPI.xxxxx   to SPI2.xxxx

works just fine with the November ESP32 but does not work (no touch seen)
with the April package.

I will test new versions occasionally to see if either of these get fixed,
but I am betting that no one cares but me.

Sam

closed 2019-06-03 by stale[bot]