ThingPulse/minigrafx #32: WaveShare42: pb with busy signal / WaitUntilIdle / IfInit

r2d2r01, 2019-01-20:
Hi I change busy pin link to gpio34 because only input and work with all others gpio.
Method  EPD_WaveShare42::WaitUntilIdle is stalled and pause process.
So I modify EPD_WaveShare42::IfInit is modified by pinMode(this->busyPin, INPUT);//OUTPUT);
Regards

squix78, 2019-01-22:
Hi. I didn't quite understand: Are you saying that we should change the driver for the 42" display to INPUT mode? And the resulting problem for you was that WaitUntilIdle was blocking for you? Thanks for your clarification...

r2d2r01, 2019-01-22:
Hi Squix exactly. When i try to launch an example to use library after first lines executed app has been stalled.
After checking some things.., for gpio pin I/O and discover a pinmode call in output mode for busy (i use gpio34 for busy because only in reading i can use it) rather than input. When changing this sens in your in IfInit, Wait function don' wait and code continue.
Thank u
Fredéric
PS: bravo for code

closed 2019-08-04 by stale[bot]