ThingPulse/minigrafx #49: Is it possible to use cyrillic fonts?

YordanYanakiev, 2019-08-12:
Is it possible to use cyrillic fonts?
 and how, if it is ?

lorol, 2019-08-17:
https://forum.arduino.cc/index.php?topic=447956

YordanYanakiev, 2019-08-18:
Thank You very much about this link, anyway I am not sure which format I should choose.
I've tried all of the possible formats with this excel config, but seems like none were working.
Or maybe i am doing something wrong. Still at least a glypse on which format is the requires should a good direction for trying combinations until it works.

lorol, 2019-08-20:
Hi, The format is Squix V3
Header:

```
const char ArialMT_Plain_10[] PROGMEM = {
  0x0A, // Width: 10
  0x0D, // Height: 13
  0x20, // First Char: 32
  0xE0, // Numbers of Chars: 224

  // Jump Table:
  0xFF, 0xFF, 0x00, 0x03,  // 32:65535
....
```

It also needed a tweak of  utf8ascii(byte ascii) function, depending where it is taken from. I already forgot these things. Something like this..

https://github.com/lorol/esp8266-oled-ssd1306/commit/e5d6f8a1f2dc2d115b47f0f06222bf54e548a7aa

BTW, I moved lately to:
https://github.com/olikraus/U8g2_Arduino
and higher level calling this driver from:
https://github.com/neu-rah/ArduinoMenu

closed 2020-03-01 by stale[bot]