ThingPulse/minigrafx #53: EPD_WaveShare_154D67.cpp: In member function 'int EPD_WaveShare154D67::Init()':

pragmantic, 2020-09-29:
EPD_WaveShare_154D67.cpp:203:47: error: 'log_v' was not declared in this scope 
     log_v("Time before HW reset: %d", millis());

EPD_WaveShare_154D67.cpp: In member function 'void EPD_WaveShare154D67::fillRam(uint8_t, uint8_t)':
 
EPD_WaveShare_154D67.cpp:370:44: error: variable-sized object 'lineBuffer' may not be initialized
   uint8_t lineBuffer[bytesPerLine] = {value};

Using library minigrafx-master at version 1.2.4 
"ILI9341_SPI.h
Demo for the buffered graphics library. Renders a 3D cube

IDE arduino 1.9   also tested on fresh daily install on another computer...

Error compiling for board NodeMCU 1.0 (ESP-12E Module).

marcelstoer, 2020-09-29:
Ouch, looks like [EPD_WaveShare_154D67.cpp](https://github.com/ThingPulse/minigrafx/blob/master/src/EPD_WaveShare_154D67.cpp) was programmed against ESP32 i.e. ESP-IDF. You may replace the `log_` calls with `Serial.printf`. PR welcome.

pragmantic, 2020-10-03:
thanx!!!

now I get:
EPD_WaveShare_154D67.cpp: In member function 'void EPD_WaveShare154D67::fillRam(uint8_t, uint8_t)':

...\libraries\minigrafx-master\src\EPD_WaveShare_154D67.cpp:370:44: error: variable-sized object 'lineBuffer' may not be initialized

ckd77, 2020-10-11:
Hi

I am in exactly the same situation.
ESPboard 2.7.4

```
/Users/regis/Documents/Arduino/libraries/minigrafx/src/EPD_WaveShare_154D67.cpp: In member function 'void EPD_WaveShare154D67::fillRam(uint8_t, uint8_t)':
/Users/regis/Documents/Arduino/libraries/minigrafx/src/EPD_WaveShare_154D67.cpp:370:44: error: variable-sized object 'lineBuffer' may not be initialized
   uint8_t lineBuffer[bytesPerLine] = {value};
                                            ^
exit status 1
Erreur de compilation pour la carte LOLIN(WEMOS) D1 R2 & mini
```

marcelstoer, 2020-10-14:
@ckd77 what you are reporting is actually the behavior discussed in #54. I am still waiting for some feedback from @pragmantic.

GurraE, 2020-10-17:
I also had the same problem, and by applying the two fixes marcelstoer mentioned, 
(replace log_ with Serial.printf and replacing linebuffer[  etc....) the problem disappeared and everything now works fine...
Thank you marcelstoer...

marcelstoer, 2020-10-17:
> replacing linebuffer[ etc....

What did you replace it with?

GurraE, 2020-10-17:
No , I replaced it with 
uint8_t lineBuffer[bytesPerLine] ;
and did not add your suggested second line 
(shame on me, as I read in the comments that you didnt like this replacement method),
 but the project "ThingPulse ESP8266 Weather Station Color"  compiled & uploaded fine in Arduino IDE after these two replacements.
(Before the replacements the software would not compile & gave the error messages  pragmantic reported..)

closed 2021-06-26 by stale[bot]