ThingPulse/minigrafx #55: Idea / Need code help for Sliders and knobs

OstertagM, 2021-01-08:
Dear Daniel,
your library is great and I want to use  for my smart home to display things.
I want to show my battery power form my solar panel if current goes into or coming out. 
I have seen how to make slider and fill squares. But how can I make an knob (16 color knob)
like this: It should be got into positive and negative from a certain start point.
I think for you or the people here this is very easy. 
[image]

lorol, 2021-01-08:
: (Joke) it is easy:  https://ej2.syncfusion.com/demos/#/material/circular-gauge/user-interaction.html   ($2500)   
Is this what you need?

OstertagM, 2021-01-08:
No I need some code for the minigrafx lib how to build this type of knob here :-)

closed 2021-09-03 by stale[bot]